friedrith/node-wifi #2: Know the active SSID?

Stan92, 2016-11-06:
Hi,
A quick question (not really an issue).
Is it possible to know on which current SSID the user is connected if she is?

And this should be an issue.
When I try to connect to SSID with a wrong password, I can't get back an error saying the password is wrong but I get connected to the last known SSID.

ThibaultFriedrich, 2016-11-06:
You are right about the current SSID it is a good idea but if I remember it i's no simple on all os. I am gonna to check it.

About your issue: what is your os ? What kind of wifi card do you use ?

Stan92, 2016-11-06:
I'm using your package with a Mac as well as a Raspberry PI (Raspbian).
Je pense que tu dois comprendre le franglais :-)

ThibaultFriedrich, 2016-11-06:
Do you have the same behavior on the two platforms ?

Stan92, 2016-11-06:
Yes it's the same behavior..
Probably there's a way to check if the connection has been successfully up, if you could get the current connected SSID.
The solution would be to compare with  the callback the "active" SSID (or an empty string if the system is not yet connected) with the SSID you want to connect to...
Of course the best of the best should be to compare the Mac Address instead of the SSID (knowing you could have several same SSID).

ThibaultFriedrich, 2016-11-07:
Do you an idea to do this in macOS ?

On linux iwconfig <interface> gives informations about the current connection:
```
wlan0     IEEE 802.11bgn  ESSID:"Elqui"  
         Mode:Managed  Frequency:2.412 GHz  Access Point: C8:3A:35:35:DD:28   
         Bit Rate=300 Mb/s   Tx-Power=22 dBm   
         Retry short limit:7   RTS thr:off   Fragment thr:off
         Power Management:on
         Link Quality=44/70  Signal level=-66 dBm  
         Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
         Tx excessive retries:0  Invalid misc:56   Missed beacon:0
```

So I should be able to get:

* ssid
* mac
* frequency
* signal_level

but there is no security information.

ThibaultFriedrich, 2016-11-07:
There is a branch called #dev with everything you need, tell me if it is enough for you and I will merge it if it is the case.

Stan92, 2016-11-07:
For mac, this is the command : ifconfig 
and I get 
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384
	options=1203<RXCSUM,TXCSUM,TXSTATUS,SW_TIMESTAMP>
	inet 127.0.0.1 netmask 0xff000000 
	inet6 ::1 prefixlen 128 
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1 
	nd6 options=201<PERFORMNUD,DAD>
gif0: flags=8010<POINTOPOINT,MULTICAST> mtu 1280
stf0: flags=0<> mtu 1280
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	ether 10:40:f3:7b:4d:0e 
	inet6 fe80::bb:ee88:8865:62fa%en0 prefixlen 64 secured scopeid 0x4 
	inet 192.168.0.22 netmask 0xffffff00 broadcast 192.168.0.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
en1: flags=963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX> mtu 1500
	options=60<TSO4,TSO6>
	ether b2:00:1a:fc:99:a0 
	media: autoselect <full-duplex>
	status: inactive
p2p0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> mtu 2304
	ether 02:40:f3:7b:4d:0e 
	media: autoselect
	status: inactive
bridge0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=63<RXCSUM,TXCSUM,TSO4,TSO6>
	ether b2:00:1a:fc:99:a0 
	Configuration:
		id 0:0:0:0:0:0 priority 0 hellotime 0 fwddelay 0
		maxage 0 holdcnt 0 proto stp maxaddr 100 timeout 1200
		root id 0:0:0:0:0:0 priority 0 ifcost 0 port 0
		ipfilter disabled flags 0x2
	member: en1 flags=3<LEARNING,DISCOVER>
	        ifmaxaddr 0 port 5 priority 0 path cost 0
	nd6 options=201<PERFORMNUD,DAD>
	media: <unknown type>
	status: inactive
utun0: flags=8051<UP,POINTOPOINT,RUNNING,MULTICAST> mtu 2000
	inet6 fe80::ee24:73ff:7aee:3626%utun0 prefixlen 64 scopeid 0x8 
	nd6 options=201<PERFORMNUD,DAD>

Thanks.. I'll make a try during the week, and let your know.

ThibaultFriedrich, 2016-11-10:
The problem with the command `ifconfig` is that it doesn't give the ssid of the wifi network, the frequency nor the signal level.

vshymanskyy, 2017-01-25:
Also interested in this feature! Thx!

thibaultfriedrich, 2017-02-06:
Do not hesitate to propose a PL

thibaultfriedrich, 2017-03-26:
Sorry for the late, but this feature has been added in the last version 1.2.4 available on npm now.

thibaultfriedrich, 2017-03-26:
Please look the new function `getCurrentConnections`

closed 2017-03-26 by thibaultfriedrich