friedrith/node-wifi #3: Not working  on Raspberry pi linux OS

DaniyalGeek, 2016-11-13:
only showing empty array

ThibaultFriedrich, 2016-11-13:
Yes, after trying to resolve issue #2, I have notice this bug. It seems to come from `nmcli` on raspbian. I don't know if it is a misconfiguration of wifi on rapsbian or a bug of `nmcli` on raspbian. I am investigating.

paul-em, 2017-01-21:
why was this issue closed? was a solution found? i still have the issue... :/

thibaultfriedrich, 2017-01-21:
@DaniyalGeek closed this issus himself. Don't know why too. This issue still remains. I don't know how to use `nmcli` on raspbian on raspberry pi. I already lost more than one day to try to understand why, without results. Unfortunately, Raspberry pi is not a priority (this project is managed by my company). But if you find a solution, do not hesitate to pull request, I will be happy to integrate it.

Which version of the raspberry pi do you use ? version 3 or an other one with dongle ?

michael-palmes, 2017-06-13:
Just got  [node-wifi](https://github.com/elqui-app/node-wifi/) up and running on a Raspberry Pi 3 (Raspbian Jessie) and an Odroid XU4 (DietPi) . Below are the steps I followed if it helps:

- `sudo apt-get install network-manager`
- `sudo nano /etc/NetworkManager/NetworkManager.conf`
- Change "managed=false" to "managed=true"
- `sudo service network-manager restart`
- `sudo nano /etc/network/interfaces`
- Comment out any wlan interfaces. For example on Raspberry Pi 3:
[image]

- `sudo reboot`
- `nmcli dev wifi connect 'wifi_ssid_here' password 'wfi_password_here'`

Your node app using [node-wifi](https://github.com/elqui-app/node-wifi/) should now work correctly.

thibaultfriedrich, 2017-06-18:
I have just tried on raspbian and it's working very well. Thx. So `node-wifi` works officially on raspbian. Except that I have to use the last command with as root. However I don't want that `node-wifi` users run their scripts as root because it is too dangereous.
Do you know how to setup nmcli to make it working as standard user ?

openedhardware, 2017-08-04:
@michael-palmes 
If I comment out any **wlan** interfaces, the PIXEL wifi interface on the top-right corner does not work.
Any idea?

thibaultfriedrich, 2017-08-13:
Commenting the interfaces in the config file make them not managed anymore by network manager so it is normal (even if it is really confusing). Since most of raspberry pi uses are without user interfaces, it is not really a problem. Moreover when I install a user interface on my linux (gnome, etc...) the setup proposed by @michael-palmes seems to be not necessary.

mmeendez8, 2017-09-19:
Have you found any solution to avoid the root use ?

friedrith, 2017-09-19:
Unfortunately no, I think it is related to a a sudo configuration. `node-wifi` seems not to be the problem. Some users have experienced the same bug on standard linux distributions.

jarrodbell, 2018-02-23:
Any solution for this?

jarrodbell, 2018-02-23:
`sudo chmod u+s,a-w /usr/bin/nmcli`
This is working for my needs.

Taken from here:
https://unix.stackexchange.com/questions/364250/run-select-shell-commands-as-root-without-giving-the-user-sudo-permisions

friedrith, 2018-02-26:
Ok thanks I will put this in the readme because it is very easy to apply. However, I think it would be safer to user sudoer file maybe.

zimbora, 2018-12-02:
After doing something similar to what was discussed here, I noticed that when using "config.iface" parameter
the first line returned from the exec command was the own parameter and it was leading to a crash in 'for' loop because fields var will only have one position and the next code, consequently fails:

      ssid: fields[1].replace(/\&\&/g, ':'),
      bssid: fields[2].replace(/\&\&/g, ':'),
      mac: fields[2].replace(/\&\&/g, ':'), // for retrocompatibility with version 1.x
      mode: fields[3].replace(/\&\&/g, ':'),
      channel: parseInt(fields[4].replace(/\&\&/g, ':')),
      frequency: parseInt(fields[5].replace(/\&\&/g, ':')),
      //signal_level: networkUtils.dBFromQuality(fields[6].replace(/\&\&/g, ':')),
      security: fields[7].replace(/\&\&/g, ':') != '(none)' ? fields[7].replace(/\&\&/g, ':') : 'none',
      security_flags: {
        wpa:  fields[8].replace(/\&\&/g, ':'),
        rsn: fields[9].replace(/\&\&/g, ':'),
      }

This is very simple to solve. I solve that by changing the line:
if (lines[i] != '') {
to 
if (lines[i] != '' && lines[i].includes(":")) {

I am using this lib in several platforms with success but raspbian took me 2 hours to set up nmcli and figure out this issue.
If you could commit it I would appreciate :)

filipppp, 2018-12-18:
@zimbora where is the path to this file?

filipppp, 2018-12-18:
@zimbora changed the src/linux_scan.js file with your if, still doesn't work

filipppp, 2018-12-22:
@zimbora I don't use the script to connect to a WIFI Network, i use it for getting the network signal strength. I use the .getCurrentConnections() method, but it always returns an empty array. This error only appears at our raspberry pi, my linux machine and windows machine work fine with it.

I'm using iwget command to get the strength

zimbora, 2018-12-22:
I will implement a function with this method for raspy and see what happens. Then I give you a feedback. My previous post only fix scanWifi function

zimbora, 2018-12-22:
@filipppp I allready implemented a function calling getCurrentConnections method and it working well. I am reading rssi through it.
If you explain your error maybe I could hep you

filipppp, 2018-12-22:
@filipppp What do you mean you implemented a function? i use the .getCurrentConnections() function from the Project itself.

Btw there is no error the currenConnections Parameter is just empty.

drewcovi, 2019-02-27:
@filipppp @zimbora @friedrith the struggle is real. got this up and running on balena, and if you provide the iface option, you do indeed get that name concatenated to the list of results (broken by \n)

I've added a pull request to put this behind us.

drewcovi, 2019-03-06:
@friedrith can we get this pulled?

https://github.com/friedrith/node-wifi/pull/66

closed 2019-09-02 by friedrith

tiznull, 2020-06-01:
@michael-palmes - Thank you for your notes! This helped me a lot. I had to do a few more things to get it to work properly on RASPBIAN version "STRETCH".  I just had to disable DHCP and also install network-manager-gnome. It is important to note that the device will lose internet connection after you make this change. This will be fixed once you add a new connection using node-wifi or the built-in "wifi" cli or nmcli. If you scripted your app properly then this is not an issue and you should be able to add the first connection under the design.

```
#> sudo apt-get install network-manager network-manager-gnome
#> sudo nano /etc/dhcpcd.conf
    # This disables DHCP on the WLAN config
    denyinterfaces wlan0
#> sudo nano /etc/NetworkManager/NetworkManager.conf
    # Change "managed=false" to "managed=true"
reboot
```

Thanks for consolidating these notes.