friedrith/node-wifi #5: returning SSID without space.

DaniyalGeek, 2016-12-25:
when i request for available wifi networks than package exclude space in SSID.

ThibaultFriedrich, 2017-01-15:
This line has beed written by an intern. Yes indeed it is quite weird. Maybe it is because there are space between and after SSID. Regex would be better.

ThibaultFriedrich, 2017-01-15:
On linux the stdout of nmcli looks like : 
*:                                       
DBUS-PATH:                              /org/freedesktop/NetworkManager/AccessPoint/1095
SSID:                                   Livebox-01ED
SSID-HEX:                               4C697665626F782D30314544
BSSID:                                  30:7C:B2:18:01:ED
MODE:                                   Infra
CHAN:                                   6
FREQ:                                   2437 MHz
RATE:                                   54 Mbit/s
SIGNAL:                                 35
BARS:                                   ▂▄__
SECURITY:                               WPA1 WPA2
WPA-FLAGS:                              pair_tkip pair_ccmp group_tkip psk
RSN-FLAGS:                              pair_tkip pair_ccmp group_tkip psk
DEVICE:                                 wlp1s0
ACTIVE:                                 no
*:                                       
DBUS-PATH:                              /org/freedesktop/NetworkManager/AccessPoint/1068
SSID:                                   SFR_B5C0
SSID-HEX:                               5346525F42354330
BSSID:                                  24:95:04:99:B5:C4
MODE:                                   Infra
CHAN:                                   1
FREQ:                                   2412 MHz
RATE:                                   54 Mbit/s
SIGNAL:                                 29
BARS:                                   ▂___
SECURITY:                               WPA1
WPA-FLAGS:                              pair_tkip pair_ccmp group_tkip psk
RSN-FLAGS:                              (none)
DEVICE:                                 wlp1s0
ACTIVE:                                 no
*:                                       
DBUS-PATH:                              /org/freedesktop/NetworkManager/AccessPoint/1096

So there are space between labels and values. Regex would definitively be better

ThibaultFriedrich, 2017-01-15:
Ok the correction is now effective in the version of `node-wifi` on npm. Thanks for the help.

closed 2017-01-15 by ThibaultFriedrich