friedrith/node-wifi #6: cannot find module "child_process"

Coder-Ashish, 2017-01-16:
actually i start using node-wifi javascript libraries in angular 2 but i am getting error cannot find module "Child_process" so can you help me how i can use this library in angular 2.
urgent

ThibaultFriedrich, 2017-01-16:
Euh, `node-wifi` is a node.js module so it cannot be used for fronted in browser. Even with some help like [browserify](http://browserify.org/), I doubt it works because `node-wifi` uses command lines.

Coder-Ashish, 2017-01-16:
then how we will run it in angular

ThibaultFriedrich, 2017-01-16:
You cannot. It is not intended for this. This module is useful for node scripts or desktop applications in nodejs. Maybe you could explain me what is the use case so I could help you.

Coder-Ashish, 2017-01-16:
i want to create app which can be display the wifi devices near you....

ThibaultFriedrich, 2017-01-16:
What is your framework ? Electron, node-wifi ?

Coder-Ashish, 2017-01-17:
ionic 2 framework and first i want to test this feature on browse.

Coder-Ashish, 2017-01-17:
node-wifi

ThibaultFriedrich, 2017-01-17:
I don't know how `node-wifi` works but you should be able to use node.js in a specific thread. The problem doesn't come from our module which is specifically adapted for node environment. That's your job to find a node environment in your framework.

Coder-Ashish, 2017-01-17:
Euh ,okkk

ThibaultFriedrich, 2017-01-17:
Sorry I made a mistake, I used several times the term `node-wifi` instead of `node-webkit`.

ThibaultFriedrich, 2017-01-17:
But I can ensure you that you cannot use `node-wifi` in the browser. It would be a significant security breach if you could because we use command lines. It is possible in some framework like electron even in the ui thread but it is not possible in simple browser.

Coder-Ashish, 2017-01-18:
so, can you provide me names of  libraries which i can use in simple browsers also for wifi. because i found some of the libraries which can run in simple browsers also.because is at this state i change my framework from ionic 2 to electron is quiet difficult and time consuming

ThibaultFriedrich, 2017-01-18:
Sorry guy but I don't have the time to do your job. This module works very well for full node environment (I thought it will obvious because of the name of the module) for Windows, MacOS and Linux (but not Android or iOS). For other environments try to find something else because we don't support them. I doubt you can find a wifi library working in a simple browser because as I already told you : "It would be a significant security breach" but if you already found one working with ionic, perfect use it.

closed 2017-01-18 by ThibaultFriedrich

Coder-Ashish, 2017-01-18:
Ehhu, ok