friedrith/node-wifi #7: Trouble connecting to networks on debian

tslater, 2017-02-11:
I get an error `Error: Failed to add/activate new connection: (32) 802-11-wireless-security.psk: property is invalid`, which is the same error I get if I try and connect via `nmcli dev wifi con "MY_SSID" password "MY_PASSWORD".
`

thibaultfriedrich, 2017-02-12:
Please give your version of debian. 
`nmcli` seemed to be good cli interface a priori but it is quite mis configured in some distributions.

tslater, 2017-02-13:
Does this work on some distros? I'm on 8.3
```
Distributor ID: Debian
Description:    Debian GNU/Linux 8.3 (jessie)
Release:        8.3
Codename:       jessie
```

thibaultfriedrich, 2017-02-14:
It works in Ubuntu but not on debian nor raspbian. It is a issue to resolve but I would like to resolve it with keeping nmcli instead of using another command line tool. I already tried a quite long time but `nmcli` is not well configured in few distributions.

thibaultfriedrich, 2017-02-14:
If you have any idea how to manage it on debian, I will integrate in the project without a doubt.

tslater, 2017-02-14:
I've been looking into it and haven't found a way with nmcli. I'll keep
looking and let you know if I find anything!
ᐧ

tslater, 2017-02-23:
I have an update here. nmcli works if I do:

`nmcli dev wifi con "MY_SSID" password 'MY_PASSWORD'` instead of `nmcli dev wifi con "MY_SSID" password "MY_PASSWORD"`. I have a dollar sign in my password. I wonder if that is related?

tslater, 2017-02-23:
It also works if I add a \$, must be bash thinking that $ is a variable name? Maybe we need something like this: http://stackoverflow.com/a/7685469/1629323. What do you think @thibaultfriedrich ?

tslater, 2017-02-24:
Calling the escape function on my string before connecting fixes it. Is that something we'd want to add to the code itself?

thibaultfriedrich, 2017-03-26:
Sorry for the absence, do not hesitate to propose a pull request, if it works for you, I will test on our machines before merging it.

thibaultfriedrich, 2017-03-26:
In fact, I will to it today.

thibaultfriedrich, 2017-03-26:
The following function doesn't work at all for me. May you be more accurate of which function works for you.

```
var escapeShell = function(cmd) {
    return '"'+cmd.replace(/(["\s'$`\\])/g,'\\$1')+'"';
};
```

Moreover, I already use ' instead of " for password and ssid.

closed 2019-09-02 by friedrith