friedrith/node-wifi #10: Only work with sudo in linux

sn0opr, 2017-03-30:
I can't connect to wifi without sudo permission on Ubuntu 14.04, here is the message I get : 

> Error: Failed to add/activate new connection: (32) Not authorized to control networking.

thibaultfriedrich, 2017-03-31:
Quite weird! It worked well with my Ubuntu. Unfortunately, I have recently switch to Fedora so I cannot test anymore on Ubuntu. Try to use directly `nmcli` in command line to check if you need sudo for it : `nmcli -f all -m multiline dev wifi list iface <interface>`.

If it is the case, you may look like in your graphical settings, if all users are authorized to connect wifi. Otherwise, you can update the file sudoers in order to authorize the command `nmcli` for everybody without password.

thibaultfriedrich, 2017-07-20:
But in the same it is totally against all security rules to start a node process with root priviledges so we should find an other solution. Moreover, I don't need to use my root password to connect to a new wifi network with a UI like gnome for example so it must be a way to give the privilegdes to simple user to create connection. It be related to sudoers file.

logidelic, 2017-12-21:
FWIW, I'm currently testing with Ubuntu 17 and my process does not need root priviledges for node-wifi to work correctly...

closed 2019-09-02 by friedrith