friedrith/node-wifi #14: Can't connect on Windows 10 with Realtek RTL8191SU Wireless LAN 802.11n

jjtbsomhorst, 2017-08-09:
As the subject states. This does not seem to be working on Windows 10. I can scan for networks but when I connect to one I directly recieve the connected message but in fact the system is not connected at all. No errors are returned.

thibaultfriedrich, 2017-08-13:
I don't know the wifi card "Realtek RTL8191SU Wireless LAN 802.11n". Is it an internal wifi card or an external ?

Moreover, Microsoft has totally rewritted Wifi layer for Windows 10 so I can understand that it doesn't work very well on this OS. Unfortunately I don't have windows 10 to debug. I can give the commands to tests:

1. Connect your Windows 10 computer to a wifi network with the UI
2. Open Windows Console (type **cmd** in the windows research bar in the startup menu)
3. Enter command `netsh wlan show profiles`
4. You should find a profile corresponding to your current network (most of the time the profile has the exact same name as the wifi network ssid)
5. Disconnect your computer from the wifi network with the UI
6. Enter command `netsh wlan connect name=<profile name> ssid=<ssid name>`. If you want more information about this command type `netsh wlan connect` without parameters.
7. Check if the connection has been established

I can understand that scan works but not connection because connection is largely more complex on windows than scan. With all these commands I should be enable to better locate the issue.

jjtbsomhorst, 2017-08-13:
Hi, I just completed the steps you posted and when I do this I correct get an active connection with the network of preference

closed 2017-12-20 by friedrith