friedrith/node-wifi #17: NetworkUtils.frequencyFromChannel function is not accurate.

obermillerk, 2017-08-18:
See [this wikipedia article](https://en.wikipedia.org/wiki/List_of_WLAN_channels).

There are channels that you don't include and also different frequencies associated with the same channel on different bands (though it looks like they're mostly inconsequential since they aren't allowed in most places?).  I should note I encountered this problem since the network I use (5GHz) gave me a channel of 149 on Windows.

thibaultfriedrich, 2017-08-18:
You are totally right. For now, it use European norms. You are totally free to propose a Pull request to manage these other places. So, it could be useful to add a settings defining the current place.

Otherwise, I think I will do it during the next months.

obermillerk, 2017-08-18:
The frequency is not particularly important to me, so I think I'll let you handle it how you see fit.  Just wanted to let you know.

thibaultfriedrich, 2017-08-18:
Ok.

friedrith, 2017-11-01:
Hi,
I have pushed a new version available on master branch (but not on npm yet) which improves the way to manage frequencies. It manages much more frequencies and add channels to the result (just in case the frequencies are still wrong for you). Can you check if it is ok for you ?

friedrith, 2017-12-20:
The new version has been published on npm so do not hesitate to try it and tell me it is ok for you.

closed 2017-12-20 by friedrith