friedrith/node-wifi #19: connect function no check the SSID, then no return when the SSID is wrong

Manu1400, 2017-09-10:
1/ On Mac, install (the last version of) node-wifi with command npm install node-wifi
2/ In Hyper (a terminal), run node
3/ Copy paste the code
`var wifi = require('node-wifi');

// Initialize wifi module
// Absolutely necessary even to set interface to null
wifi.init({
    iface : null // network interface, choose a random wifi interface if set to null
});
4/ Connect your computer to your wifi, with GUI.
`
5/ Copy paste the other code
`// Connect to a network
wifi.connect({ ssid : "FreeWifi", password : ""}, function(err) {
    if (err) {
        console.log(err);
    }
    console.log('Connected');
});`
6/ Wait few seconds.

node-wifi try to connect FreeWifi SSID (it's a free wifi not secured in France from "Free" popular company). But failed. Mac reconnect your favorite wifi SSID. And node-wifi say "Connected" because no check the SSID connected.

Expectation : throw an error.

friedrith, 2017-09-15:
Bonjour,
J'avoue ne pas avoir testé `node-wifi` sur un freeWifi.
Est ce que tu pourrais tester les manipulations suivantes ?

1. lancer la commande `ifconfig` et vérifier que l'interface **en0** existe bien.
2. lancer la commande `networksetup -setairportnetwork "en0" "<ssid>"  "<password">

Pourrais-tu afficher le résultat ici ?

Manu1400, 2017-09-16:
Bonjour,
En somme, le problème n'est pas FreeWifi, mais tout simplement que si tu n'arrive pas à te connecter à un réseau pour une raison ou une autre, Mac OS te reconnecte automatiquement à ton réseau wifi (en suivant l'ordre de la liste de préférence de réseau wifi).

friedrith, 2017-09-17:
C'est tout à fait possible mais j'avoue que `node-wifi` devrait emettre un message d'erreur même si c'est jamais très simple avec les commandes wifi.

friedrith, 2017-10-02:
Je viens de tester avec la configuration suivante : 

- macOS 10.12.5
- FreeWifi

Et cela fonctionne parfaitement avec le code qui tu as mis ci-dessus. Cela peut dépendre de ta configuration réseau (si échec automatiquement se reconnecter à un autre réseau ?). Mais j'avoue ne pas trop m'y connaître en mac sur cela. A tout hasard, je vais améliorer la stabilité en améliorant les gestions des retours d'erreurs dans `node-wifi`.