friedrith/node-wifi #21: Connection on Peap access point

lyoko17220, 2017-10-10:
It's only possible to connect with access point that needed a simple password. 

That's a problem for Student and users of Eduroam for example

friedrith, 2017-10-13:
Hi,
Can you describe me how you authenticate you on a Eduroam wifi ?
I think you must type your login and password on a web page if I remember well my school years.

If it is the case, `node-wifi` cannot do anything because each wifi portal uses the protocole it wants. It is impossible to manage all of them.

lyoko17220, 2017-10-13:
Hi,

My configuration on `/etc/wpa_supplicant/wpa_supplicant.conf` file :

`network={
       ssid="eduroam"
       priority=1
       proto=RSN
       key_mgmt=WPA-EAP
       pairwise=CCMP
       auth_alg=OPEN
       eap=PEAP
       identity="username"
       password="password"
       phase1="peaplabel=0"
       phase2="auth=MSCHAPV2"
}`

friedrith, 2017-10-14:
You don't use the web interface to connect your computer ?

lyoko17220, 2017-10-14:
No, only the configuration file is only needed

friedrith, 2017-10-14:
Ok and How does it work on windows or macOS ?

lyoko17220, 2017-10-14:
When you connect to the access point username/password is require. 

You can look for example [how to connect to Eduroam access points](http://www1.uwe.ac.uk/its/usingitatuwe/connectingtowi-fi/configureyourdevice.aspx) because all Eduroam WiFi networks use Peap protocol

friedrith, 2017-10-23:
On windows, `node-wifi` uses `netsh` command. On linux, it is `nmcli` and on macOs it is `airport`. Do you know I could manager peap with these commands because I didn't see any mention to peap protocol on their documentation ?