friedrith/node-wifi #22: Connect to hidden SSID

sliard, 2017-10-31:
When I try to connect to a secret SSID, I have the error :
`Error: No network with SSID 'name' found`

I think it's because nmcli need information about wifi before connection (like you can [read here](https://stackoverflow.com/questions/35476428/how-to-connect-to-hidden-wifi-network-using-nmcli)) :
```
nmcli c add type wifi con-name <connect name> ifname wlan0 ssid <ssid>
nmcli con modify <connect name> wifi-sec.key-mgmt wpa-psk
nmcli con modify <connect name> wifi-sec.psk <password>
nmcli con up <connect name>
```

friedrith, 2017-11-01:
Hi,
First of all, thank you for remarks.

I didn't know about secret SSID.What I understand with your explanation is that wifi network with secret ssid don't diffuse security protocols and so a computer cannot be easily connected using `node-wifi`. On windows, you must use specials commands to manually setup security protocoles in the profile.

My questions are no:
- Are my interpretations correct ?
- Do you know how create a secret ssid in order to reproduce your problem ?
- Are you ready to implement this feature in `node-wifi`. Please do not begin it today because I am going to release the version 2.0 of `node-wifi` in the next days.

sliard, 2017-11-01:
Hi @friedrith 

Yes, your interpretations are correct.
I think that any wifi router can create hidden SSID (like [this netgear](https://kb.netgear.com/30230/How-to-hide-your-NETGEAR-wireless-router-SSID-network-name))
In this case you have to disable "SSID Broadcast"  

I worked on it yesterday and had a lot of problems because nmcli support "add" and "modify" command only since Ubuntu 16 but my device use Ubuntu 14.
For the moment the only way I found need to declare the SSID with graphical interface before use it with a "nmcli up" command.

And yes I fork your project to start a PR but can't find the right way for the moment.
So I can way for your 2.0 version :)

Samuel

maxhellwig, 2022-09-20:
Hello, 
I'm working on an implementation to handle hidden wifis on Windows:
https://github.com/friedrith/node-wifi/pull/183
An PR is open but it failes sometime during connection attempt to the hidden wifi. I suspect a timing issue during profile writing and netsh commands.

Kind regards,
Max