friedrith/node-wifi #27: Error while getCurrentConnections()

haloszymek, 2017-11-27:
Hey guys, 
I need help with getCurrentConnections function. I need to get Promise back, I try to do if this way:

```
wifi.init({ iface : 'wlp3s0' });

wifi.getCurrentConnections().then(function (current) {
  console.log(current)
}).catch(function (error) {
  console.log(error)
})
```
I have this error instead:

```
/home//JavaScript/sandbox/main.js:8
wifi.getCurrentConnections().then(function (current) {
                            ^

TypeError: Cannot read property 'then' of undefined
    at Object.<anonymous> (/home/john/JavaScript/sandbox/main.js:8:29)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
    at Function.Module.runMain (module.js:676:10)
    at startup (bootstrap_node.js:187:16)
    at bootstrap_node.js:608:3
```

But when I try to use it in my project, I get different error:

```
{ Error: Command failed: nmcli --terse --fields active,ssid,bssid,mode,chan,freq,signal,security,wpa-flags,rsn-flags,device device wifi ifname wlp3s0
Error: 'device wifi' command 'ifname' is not valid.

    at ChildProcess.exithandler (child_process.js:275:12)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:554:12)
  killed: false,
  code: 2,
  signal: null,
  cmd: 'nmcli --terse --fields active,ssid,bssid,mode,chan,freq,signal,security,wpa-flags,rsn-flags,device device wifi ifname wlp3s0' }
```

What can goes wrong?

micah-boswell-aquabotix, 2017-12-10:
The problem is that nmcli expects a command with format:
nmcli device wifi [list [ifname ifname] [bssid BSSID]]
but the code is leaving out 'list' which it needs in order to have a valid command.

friedrith, 2017-12-20:
Thx for the PR.

closed 2017-12-20 by friedrith