friedrith/node-wifi #28: getCurrentConnections does not work on mac

felixgalindo, 2017-11-29:


peterbe, 2017-12-06:
I too get nothing:
```javascript
wifi.init({
    iface : null // network interface, choose a random wifi interface if set to null
});

wifi.getCurrentConnections(function(err, currentConnections) {
  if (err) {
    console.error(err);
    throw err
    // console.log(err)
  }
  console.log('CURRENT CONNETIONS');
  console.log(currentConnections)
})
```
Using the latest version.

peterbe, 2017-12-19:
To anybody else ending up here and being too eager to wait for a fix, here's what I did:
https://github.com/peterbe/howsmywifi/blob/2c479bcbd344fce0b9a77b9fe972a38e24b78262/index.js#L30-L68

I only have a mac so I wouldn't even be able to test `node-wifi` on non-macs. The `spawn()` solution works for mac. I was going to use that as a fallback.

xdadda, 2017-12-19:
As mentioned no need for a fix, just call the Promise as explained above. 
And it will work on all platforms.

peterbe, 2017-12-19:
Sweet! Thanks. [Removed my hack](https://github.com/peterbe/howsmywifi/commit/c0451eb0984e959e9c43885b88096a7f13bc51db)

friedrith, 2017-12-20:
Sorry for the late. I have pushed a new version that corrects the bug of callback on macOS.

closed 2017-12-20 by friedrith