friedrith/node-wifi #31: What's the reason for not auto-connecting to open networks?

paumoreno, 2017-11-30:
I'm wondering why, at least on Windows (i haven't tested the behaviour on other OSs), when a connection is established to an open newtork, the `connectionMode` is set to `manual`. This prevents the machine from connecting to the chosen network again after a restart.

Is it a security reason? Does it happen on Linux and Mac?

friedrith, 2017-12-02:
In fact this library is made to manage wifi connections without human interaction. And it is better to do it in manual mode. It could be a improvement to do.

friedrith, 2017-12-20:
It could be something useful to add into the options in the connect function like this:

```js
wifi.connect({
    ssid: 'ssid',
    password: 'password',
    auto: true|false,
})
```

However, I think that could be useful to find how to manage this auto connection behavior for all OS before integrating it in `node-wif`.

stonegray, 2018-04-12:
It would be a huge confidentiality and availability problem. 

Availability wise, it's quite possible you'll hit a network with a captive portal and not be able to access the web. How can you tell if you've connected to a useful network?

From a security perspective it's a nightmare. You can fingerprint the device, MITM your device, and more.  Mitigating this is way outside the scope of this library. 

If you want to do this you totally can, but beware of the above.

friedrith, 2018-05-13:
Totally right! But since it is available with user interface, I don't think it is really a problem. Moreover this option should be set to false by default. 
Whatever, I am waiting to have this feature on all OS before releasing it.