friedrith/node-wifi #39: Parser bombs, callback not called

lordbah, 2018-03-07:
It appears the fix for issue 26 has caused a new variant of issue 20 where the parser bombs but does not call the scan callback with the error. I think it's because the try/catch inside the exec() callback in windows-scan.js was removed. In my case the result is the process exits, I'm not sure if that's always true. Here's the current exception:

```
...\node_modules\node-wifi\src\windows-scan.js:47
    network.mac = networkTmp[4].match(/.*?:\s(.*)/)[1];
                               ^

TypeError: Cannot read property 'match' of undefined
    at parse (E:\KPM\SrvWebs\vgr-system-server\node_modules\node-wifi\src\window
s-scan.js:47:32)
    at E:\KPM\SrvWebs\vgr-system-server\node_modules\node-wifi\src\windows-scan.
js:33:27
    at ChildProcess.exithandler (child_process.js:262:7)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at maybeClose (internal/child_process.js:927:16)
    at Socket.stream.socket.on (internal/child_process.js:348:11)
    at emitOne (events.js:115:13)
    at Socket.emit (events.js:210:7)
    at Pipe._handle.close [as _onclose] (net.js:545:12)
```

parse() throws the exception. The exec() callback function which called parse() doesn't have a try/catch, the try/catch you see is for the exec() call itself.

friedrith, 2018-05-13:
Sorry for the delay. I am not sure to understand the reason. The try catch outside the exec call should handle this error.  I am investigating.

lordbah, 2018-10-25:
It happened again today on Windows 10. Sometimes "netsh wlan show networks" will encounter a network for which it prints the SSID, Network type, Authentication, Encryption, but nothing else. So parse() throws an exception trying to parse the MAC, and in the callback function passed to exec() from scanWifi() there is no try/catch, so the process is terminated. The try/catch around the call to exec() doesn't help, because the exec() function has already returned.

friedrith, 2018-10-27:
Hi, 
In order to improve the stability of this library, I think it would be better to create unit tests and have example of netsh standard output to reproduce this kind of situtation. Can you provide an output related to your situation to include it in the unit tests ?

lordbah, 2018-11-05:
I haven't caught it in this state. Will update if I do.

abdurrahmanekr, 2019-01-18:
I was having this problem before this issue. I opened a #24 to solve the problem. But since the code was deleted, I had to switch to the old commit. I'll be following this issue.