friedrith/node-wifi #47: Is it possible to set up event listeners for changes?

blahsd, 2018-05-07:
So have a function triggered, for instance, when the connection is lost? Or do I have to ping it constantly to monitor if there are any changes?

friedrith, 2018-05-13:
Hi,
It could very useful but I suggest you to do it by your self on the top of `node-wifi`. Because neither linux, macOS or Windows propose simple CLI commands to get event listener. So `node-wifi` should have a system of very stable pooling to handle this kind of feature which is very particular for your case.

To be honest, that something I built in my company where I created this project but it was so painful so have something stable (you have a lot of wrong positive and wrong negative) that I didn't want to integrate it in `node-wifi`. Sadly, I didn't save the code so it couldn't help you.

It think it could be a new project built on the top of `node-wifi` or built using the C api of each OS maybe (it might be more efficient).

closed 2018-05-13 by friedrith

vallabhvrao, 2018-10-30:
Can't `nmcli` monitor mode be used to watch changes ?
`nmcli -p m`
It doesn't return after it goes in monitor mode, will node still be able to get the output even if the control is not returned ?