friedrith/node-wifi #52: How do I contribute to the source code?

joanferns00, 2018-08-29:
Hi, 
      I'm really interested in your code and what you are doing. I see that you wanted help on your npm page. Please let me know how to get started. Thanks!

friedrith, 2018-09-01:
Hi,
I am pleased to hear so you want to help. Thank you for proposing yourself. I admit I would definively need help. I don't have a lot of time for this project even if I know it may has a great future. 

Because of this lack of time, I never had the chance to upgrade node-wifi to its next level but with some help I think it is easier to achieve.

The next plans I have now are:
* reimplement the full library using Ecmascript 2015 (or event 2017+)
* add tests
* correct most of issues 

What do you think about it ?
And thank you again for your proposition. It is very nice from you.

verneck, 2018-10-16:
some of the items mentioned above have already been done? @friedrith

verneck, 2018-10-16:
- [ ] reimplement the full library using Ecmascript 2015 (or event 2017+)

 - [ ] add tests

 - [ ] correct most of issues

ShayD16, 2019-01-30:
@friedrith What's the status of this now? I would be interested in contributing to the ECMA2015 migration

friedrith, 2019-01-30:
Hi @ShayD16,

I will push a first draft of the new version very soon. It will add Ecmascript 2015 and also unit tests but it takes some time. I will add a comment on this thread once it's done.

blackshady, 2020-05-13:
Hi, good day @friedrith Please what is the status of this project? would love to contribute.

friedrith, 2020-08-11:
Oups @blackshady  sorry I didn't see your message since it was on an old issue. I am currently working on a strong refactorization of node-wifi in order to make it easier to test automatically and also to rewrite it with ES6.