friedrith/node-wifi #53: Airport data parsing fails when a channel >99 is used

matthiaslau, 2018-09-08:
When there is a wifi in reach using a channel >99 the parsing the airport output with the title indexOf will result in faulty data. E.g.

```
                       A Wifi 58:b8:f3:85:55:0e -84  44      Y  DE WPA2(PSK/AES/AES)
                 Another Wifi 5 GHz 58:5e:12:48:35:24 -53  100,+1  Y  DE WPA2(PSK/AES/AES)
```

will result in 

```
ssid: "Another Wifi 5 GHz 58"
bssid: ":5e:12:48:35:24 -5"
signal_level: "3 10"
```

It could be fixed using a regexp like this: https://github.com/ancasicolica/node-wifi-scanner/blob/master/lib/airport.js#L10

friedrith, 2018-10-11:
Hi,
Sorry for the delay. Thank for noticing this bug. I will fix it next week. I am trying to improve the stability of node-wifi and for that I need outputs from different OS. Could you please send a airport output relating to your situation ?

matthiaslau, 2018-10-29:
I am sorry, I don´t have the tested wifi around me any more. It was a 5GHz only wifi with a channel >100.