friedrith/node-wifi #61: Working when deployed?

weiwanghasbeenused, 2019-01-16:
Hi,

I'm new to Node so please forgive me if this is dumb.
I have a code that detects wifi signals around me and set up a server for another page to grab it when I ran it locally. However, when I deployed it on Heroku, It just won't work. Is it because of that part of setting up a server or...? I would appreciate if you can spend some time looking at this. Thank you!

`var data = [],
	id = [],
	signal = [],
	address = [],
	flag = [],
	fontSize = [],
	fontColor,
	position = [],
	wW = $(window).width(),
	wH = $(window).height(),
	lkk = 0;
function addDiv(i, c, parent){
	$(parent).append("<div id = '"+i+"' class = '"+c+"'></div>");
}
function randomRange(d,max){
	return (max-d)*Math.random();
}		

$.get("https://localhost:8081/getwifis", function(data){
	for(i = 0; i<data.length;i++){
		id[i] = data[i].ssid;
		signal[i] = -parseInt(data[i].signal_level);
		address[i] = data[i].mac;
		flag[i] = data[i].security;
		fontSize[i] = signal[i];
		addDiv("c"+i,"clouds blur","#container");
		var left = randomRange(signal[i]*wW/200,wW);
		var top = randomRange(signal[i]*wH/200,wH);
		$("#c"+i).css({"width":signal[i]/2+"vw","height":signal[i]/2+"vh","top":top+"px","left":left+"px"});
		$("#c"+i).append("<p>"+id[i]+"</p><p>)))"+signal[i]+"</p>");
		console.log(id[i]+", signal = "+signal[i]);
	}
	
});`

Wei

friedrith, 2019-01-17:
Hi, 

Can you format your code using code balise in markdown.

Node-wifi is not meant to be used on server since most of the time servers don't have any wifi card. And you cannot used node-wifi in the broswer since it uses child-process from node api.

closed 2019-01-17 by friedrith

weiwanghasbeenused, 2019-01-17:
I see...I guess I need to find another library. Thank you though!

Sorry about the messy code. Here's the formatted one if you're interested. Some CS student helped me with it so I'm not 100% sure about how it works.

```javascript
var express = require('express');
var app = express();
var wifi = require('node-wifi');
var parsedNetworks;
var cors = require('cors');
var ip = require('ip');
var myIP = ip.address();
var https = require('https');
var path = require('path');
var fs = require('fs');
 
console.log(myIP); 
app.use(cors())

/*wifi*/
wifi.init({
	iface :null
});
wifi.scan(function(err, networks){
	if(err){
		console.log(err);
	}else{
		// pushing data to array "parsedNetworks"
		parsedNetworks = [];
		networks.forEach(function(e){
			parsedNetworks.push({'ssid': e.ssid, 'signal_level':e.signal_level, 'mac':e.mac, 'security':e.security})
		})
		// consoling ssid of each signal
	  	for(i=0;i<networks.length;i++){
	  	  console.log(networks[i].ssid);
	  	}
	}
});
// sending wifi data at url/getwifis
app.get('/getwifis', function (req, res) {
   res.json(parsedNetworks);
})

// setting up https server
var certOptions = {
  key: fs.readFileSync(path.resolve('server.key')),
  cert: fs.readFileSync(path.resolve('server.crt'))
}

var server = https.createServer(certOptions, app).listen(8081,function () {
   var host = server.address().address
   var port = server.address().port
   console.log("Example app listening at https://%s:%s", host, port)
})
```