friedrith/node-wifi #64: Uncomplete list of AP over linux

waken22, 2019-02-21:
Hi, thanks to read the issue, i didn't found a guideline to just open an issue.
_(Currently working on a project that uses node-wifi inside an IoT device, nmcli tool, version 1.2.6)_

So I was having troubles to get an updated list of visible APs, not sure if my issue is related with network manager but I seems that i find a way to make it works using ```nmcli device wifi rescan``` before a ```wifi.scan();``` only when i did that it works, so my question is if this is a specific workaround for me or it happens to someone else?

waken22, 2019-02-21:
I'm not sure if the rescan is needed but i found something that is giving me problems and is related by the title topic.

I want to ask what is the propose of this inside linux-scan.js and linux-current-connections.js
```javascript
      if (config.iface) {
          lines.shift(); // ???? what
      }
```
I don't understand why is needed a shift here, it removes an AP from the array... and by that i'm not getting all the APs...

francoisrousseau, 2019-07-17:
Amazing find waken22! 
I have the same issue... I also had to remove the shift inside "getCurrentConnection" from file linux-current-connections.js