friedrith/node-wifi #72: random error while scanning wifi networks

bennymeg, 2019-06-16:
Hi, 
I am trying to scan for all the available networks around me, unfortonetly, most of the times I receive the following error: _"can't read property 'match' of undefined"_.
Some of the times it works currectly.

Currently the only thing I do is setting the interface to null on the init() function (so the interface is selected randomly) and invoking the scan() function.
`wifi.init({ iface: null });
wifi.scan();`

Any idea what might gone wrong?

bennymeg, 2019-06-16:
OK, I managed to find and fix the bug.
It seems that my network card sometimes return only part of the data you expect, and the code break when you try yo parse the data.

In example, it might return only those fields:
`[ 'SSID 1 : xxx',
    '    Network type            : xxx,
    '    Authentication          : xxx',
    '    Encryption              : xxx ' ]`  

I fixed the issue and I will submit a PR soon.
Benny.

friedrith, 2019-07-10:
Fix included in version 2.0.7

closed 2019-07-10 by friedrith