friedrith/node-wifi #78: Network profile is saved as wrong filename

Spitz3492, 2019-08-05:
https://github.com/friedrith/node-wifi/blob/eeaef529e95554e3abea70f2224d2aaece80bb20/src/windows-connect.js#L34

The windows connect function saves the new network profile under the filename "nodeWifiConnect.xml" but then immediately after uses the netsh command to add the profile using the filename (wifi ssid + ".xml"). This gives me the error, "stdout: 'The system cannot find the file specified.\r\n\r\n\r\n',". So line 34 should write the filename as (ap.ssid + ".xml") not "nodeWifiConnect.xml". After I made this change to the writeFileAsync and the following execCommand delete on line 46, my program connecting to the network successfully.

samuelcseto, 2019-08-05:
Hi, I have already fixed this in #76. The library later tries to delete the file named 'nodeWifiConnect.xml' So I guess the correct file name should be 'nodeWifiConnect.xml' and not the SSID.

SalatielSauer, 2019-08-06:
Hello, I believe the problem still persists in the [NPM version](https://www.npmjs.com/package/node-wifi).

samuelcseto, 2019-08-06:
The pull request is still open so that's why.

S pozdravom,
Samuel Csető

Spitz3492, 2019-08-07:
Change the filename to "nodeWifiConnect.xml" then as well on [line #40 ](https://github.com/friedrith/node-wifi/blob/eeaef529e95554e3abea70f2224d2aaece80bb20/src/windows-connect.js#L40) because that is what is causing my issue. Your pull is screwing up the master branch of this npm install.

Spitz3492, 2019-08-07:
I checked out your branch and it seems you have already fixed this issue. Please commit that change to master.

samuelcseto, 2019-08-07:
I am not the author of this repo so I can't merge into master until the author accepts my pull request #76

friedrith, 2019-08-15:
Sorry guys, I was in vacation without internet connection. I will merge this pull request #76 as soon as possible.

closed 2019-09-02 by friedrith