friedrith/node-wifi #81: Raspberry pi UNRECOGNIZED OS error

mowi22, 2019-08-22:
wifi.init thows error UNRECOGNIZED OS error running on raspbarry pi inside of a multi-container  docker-compose file running node 10 in a electron/ react application. Im building the image via balenaOS 

## Expected Behavior
Expected it to recognize linux and use linux js files

## Current Behavior
just throws an error when on the pi. Works fine on a Mac

## Affected features

- [x ] node API
- [ ] cli

## Possible Solution
a way to specify the linux os via a param

## Steps to Reproduce (for bugs)

1. 2. 3. 4.
1. install electron
3. add node-wifi to package.json
4. build with docker-compose to balenaOS / linux-arm32 
5. run on rpi
6. fails on wifi.init
7. Unrecognized OS

## Context

I'm trying to display networks to a user via a react application. Basically, show user current wifi router and allow them to switch via the app. 
## Your Environment

- OS: linux
- node-wifi version 2.0.7
- wifi card: rpi wifi

friedrith, 2019-08-26:
Hi,

Most of the time, management of wifi is quite hard inside virtual machine or docker. Are you sute your container has access to a wifi card ?

Tobias-Keller, 2020-01-17:
I got "spawn nmcli ENOENT" on an raspberry, on osx working fine

abertranb, 2020-01-21:
I got the same issue as @Tobias-Keller

Tobias-Keller, 2020-01-25:
@abertranb got it working...follow this -> https://www.raspberrypi.org/forums/viewtopic.php?t=179576

mchsdl, 2020-10-14:
@Tobias-Keller what post from the link are you refering to? I also got this problem on raspberry.