friedrith/node-wifi #83: Security vulnerability: Command Injection

SteinRobert, 2019-08-26:
## Expected Behavior
See: Do not be vulnerable to command injection

## Current Behavior
See: https://www.npmjs.com/advisories/952

## Affected features

- [x] node API
- [ ] cli

## Possible Solution

Sanitize input.

SteinRobert, 2019-08-27:
Hi @friedrith ,

I think we can solve the issue by using `execFile` instead of `exec`. I believe I can provide you with a PR for this.

However, if I clone the current `master` branch and run following command:
```bash
./bin/wifi.js --connect --ssid test --password test
```
I get 
```
/node-wifi/node_modules/chalk/templates.js:109
				throw new Error('Found extraneous } in Chalk template literal');
				^

Error: Found extraneous } in Chalk template literal
```

After adapting the chalk syntax in `bin/wifi.js` it works fine. Is that another issue?

SteinRobert, 2019-09-06:
@friedrith I know you're probably super busy, please don't get me wrong.
If you find any time to release a new version which includes #85  me and my team would really appreciate it!

friedrith, 2019-09-06:
I have juste release new version 2.0.12 that includes all changes on master branch.

closed 2019-09-06 by SteinRobert