friedrith/node-wifi #87: Scan gives some duplicate results

FinlayDaG33k, 2019-09-10:
When I use `wifi.scan()`, I get double results of some networks (albeit with different network strengths)

## Expected Behavior

Only a single entry per network in range

## Current Behavior

Multiple entries for a network in range

## Affected features

- [x] node API
- [ ] cli

## Possible Solution

Checking the scan results for duplicates and either updating or removing them.

## Steps to Reproduce (for bugs)

1. Go sit in a place with some networks around (using a laptop)
2. add following code to a nodejs project
```js
var wifi = require("node-wifi");
wifi.init({iface: null});

 wifi.scan() // here there be dupes
```
3. run nodejs project
4. while scan is running, move your laptop around a bit

## Context

It's making my results list a little mess because it has plenty of duplicates (I can count my homenetwork in the list 4 times).

## Your Environment

- OS: SolusOS 4
- node-wifi version: 2.0.12
- wifi card: something `Qualcomm Atheros AR93xx` (unknown specific card)

friedrith, 2019-09-13:
On which SolusOS is based ? Linux ? If so, please send here the standard output of following command:

```bash
nmcli --terse --fields active,ssid,bssid,mode,chan,freq,signal,security,wpa-flags,rsn-flags device wifi list
```

I suggest you to change a little the ssid and bssid for confidentiality reasons.

FinlayDaG33k, 2019-09-13:
Yes, SolusOS is GNU/Linux based (I'm running kernel `5.2.7-123.current` to be exact).
This is the output using the command you provided:
```
no:Roelofs:B8\:69\:F4\:XX\:XX\:XX:Infra:1:2412 MHz:100:WPA2:(none):pair_tkip pair_ccmp group_ccmp psk
no:Roelofs-Gasten:BA\:69\:F4\:XX\:XX\:XX:Infra:1:2412 MHz:100:WPA1 WPA2:pair_tkip pair_ccmp group_ccmp psk:pair_tkip pair_ccmp group_ccmp psk
no:LibreMesh:60\:E3\:27\:XX\:XX\:XX:Infra:1:2412 MHz:100::(none):(none)
no:Finster:B8\:69\:F4\:XX\:XX\:XX:Infra:4:2427 MHz:100:WPA2:(none):pair_tkip pair_ccmp group_ccmp psk
no:LibreMesh-59bf:B8\:27\:EB\:XX\:XX\:XX:Infra:11:2462 MHz:100:WPA1 802.1X:pair_ccmp group_ccmp 802.1X:(none)
no:Roelofs-Gasten:BA\:69\:F4\:XX\:XX\:XX:Infra:149:5745 MHz:100:WPA1 WPA2:pair_tkip pair_ccmp group_ccmp psk:pair_tkip pair_ccmp group_ccmp psk
no:Roelofs:B8\:69\:F4\:XX\:XX\:XX:Infra:149:5745 MHz:100:WPA2:(none):pair_tkip pair_ccmp group_ccmp psk
yes:Finster:B8\:69\:F4\:XX\:XX\:XX:Infra:36:5180 MHz:99:WPA2:(none):pair_tkip pair_ccmp group_ccmp psk
no:Roelofs:18\:D6\:C7\:XX\:XX\:XX:Infra:1:2412 MHz:92:WPA1 WPA2:pair_tkip pair_ccmp group_tkip psk:pair_tkip pair_ccmp group_tkip psk
```

For privacy reasons, I have only listed the networks here that I own.

After taking a look at the output myself, I noticed that the `5GHz` networks are seperate from the `2.4GHz` networks.
Could this be the cause?

friedrith, 2019-09-13:
Yes definitevely.
Everytime a network use a different frequency, it has to be emitted by a different wifi router (or at least a different antenna on a router) and so it is identified with a different **bssid**. Most of the time router use same **ssid** so that it is transparent for the users and computers switch from one to another  with transparency too. But it is not exactly the same network, you can have different security rules for each network for example. 

If you use network repeaters you can also have the same situation : several routers and so several real networks even if only one is visible in the graphical user interface.

friedrith, 2019-09-13:
I cannot see the last caracters in the bssid of the the different network **Roelofs** but I think it is really different wifi networks. They may be binded in the router so devices can talk to each other on the unified virtual network but it doesn't seem to be the same physical network.

FinlayDaG33k, 2019-09-13:
I have taken a look and yes, they do have different `BSSID` (`05` and `06` as final characters).
I have currently written my own code snippet to do some filtering to get around this.

I'll leave it here, feel free to integrate it (maybe with a configuration flag?) or leave it be (depending on how you feel like it).
It's not perfect but it seems to do the job.
```js
// Remove all duplicate networks from the array
      var networks_temp = [];
      networks = networks.filter((item) => {
        // Check whether we already have this SSID in our results
        // Add it to the networks_temp array if we don't
        let i = networks_temp.findIndex(x => x.ssid == item.ssid);
        if(i <= -1){
          networks_temp.push(item);
          return;
        }

        // Check whether the signal quality - with a small margin - is higher or not
        // Skip this network if it's much weaker anyways
        if((item.quality + 5) < networks_temp[i].quality) return;

        // Check whether new network is using 5GHz or not
        // Skip this network if it's using 2.4GHz anyways
        if((item.frequency) < 3000) return;

        // The network is 5GHz *and* has a relatively higher signal, use it insead.
        networks_temp.splice(i,1);
        networks_temp.push(item);

        return;
      });
```

Closing this issue as this behaviour is not a bug.

closed 2019-09-13 by FinlayDaG33k