friedrith/node-wifi #92: Return the network after connecting

FinlayDaG33k, 2019-09-19:
Hii there,

node-wifi has the `wifi.connect()` method but this only returns anything if it there is an error.  
When I manually tried connecting using the following command, however, I can get some more information (most notably, the interface used).  

> $  nmcli -w 10 device wifi connect REDACTED password REDACTED
> Device 'wlp2s0' successfully activated with 'REDACTED'.

Would it be an option to return the connection (things like ssid, bssid, iface etc. etc.) so we can extract some information that might be handy?  
Right now what I do is `wifi.connect()`, wait for the connection to be finished then run `wifi.getCurrentConnections()`, wait for that to finish then do some additional checks so I can get the exact interface used.

friedrith, 2019-09-19:
Hi,
This is a good idea. We only need to be careful to have this feature on all OS. I guess you are on linux. And I am not sure windows command returns the same level of information.

FinlayDaG33k, 2019-09-19:
Yes, I'm on Linux :) 
I'm already working on it though currently, I only have the interface information and not the rest of the connection.  
I'll create a PR when it's ready.

valkheim, 2019-11-02:
Hi @FinlayDaG33k 
What do you think about having a function that returns the connection status (connected, ssid, iface, dns, ...) independently from the `wifi.connect()`, so we can use it separately.
That would fits your needs but also give the ability to use it at anytime. (See https://github.com/friedrith/node-wifi/issues/97 related issue)

FinlayDaG33k, 2019-11-02:
That would indeed be a suiting alternative yes :)

valkheim, 2019-11-04:
Hi,

I figured out that network manager sometimes fails to connect (rarely happens though) with the following message: `Error: No network with SSID 'ssid' found.`. Maybe we should add the possibility to try connecting for a pre-defined amount of times before returning an error ?

digaus, 2020-01-08:
Any progress on this? I also think getting the current network information in a separate call is the better approach.

FinlayDaG33k, 2020-01-08:
I haven't worked on it lately (mainly because I've forgotten about it since the app that I built using this hasn't been of much use for me since I don't travel by public transit a lot anymore)...

I do agree that having a separate call might be a better thing as well.

digaus, 2020-01-08:
I am interested in this because of the integration of smarthome devices. Currently using Ionic Capacitor which supports Android, iOS and Electron. So I wanted to use this plugin to do the Wifi handling for a the electron application. For this I need to know the connected SSID and device IP, anyway great work so far :) might try it out later today.

FinlayDaG33k, 2020-01-08:
@digaus Never heart of Ionic Capacitor before. I will check it out since it sounds interesting for my particular app as well.  
My particular app is basically an app to make the use of captive portals a lot quicker by automating most of the stuff.
All I really needed was the mac address of the specific network card connected to the AP (in case of multiple).

But yes, this would be quite handy.

stigmat4j, 2021-06-19:
`exec('nmcli -g ip4.gateway connection show ' + wifi.ssid, (err, ip) => {
                if(err) console.info(err);
                console.info(ip);
            });`

work for me