friedrith/node-wifi #97: Test connection

valkheim, 2019-10-26:
Hi,

It would be nice to add a connectivity test.

My scenario: I'm using node-wifi at boot to achieve some wifi related tasks and I want to be sure I'm connected before doing so.

A possible implementation using NetworkManager would be to use the `nm-online` binary (see [man page](https://linux.die.net/man/1/nm-online)) or the more common `nmcli con status <connection name>`. Not sure about windows

friedrith, 2019-10-29:
Hi,
Does this proposal may be related to #92 ?

vinicius5581, 2020-12-24:
I have a similar ask and after reading #92 I don't think that is the solution I'm looking for. 

I would like to make a call as `wifi.status` and get on the response an object containing the status either connected or not connected and if connected, if details about the current connection.

Thoughts?