friedrith/node-wifi #101: Linux - Scan doesn't rescan but list networks

bfamchon, 2019-12-11:
#  Expected Behavior

In some case, wifi list doesn't update automatically and we need to rescan networks manually using linux cmd `nmcli wifi rescan`.

## Current Behavior

Actually with node-wifi, usage of `wifi.scan()` do a `nmcli wifi list`

## Affected features

- [X] node API
- [ ] cli

## Possible Solution

Change `list` argument to `rescan` at this line:
https://github.com/friedrith/node-wifi/blob/master/src/linux-scan.js#L14

## Context

Changing this line will do a rescan instead of a list, cover more use case and fit with the name 'scan'

Joker9090, 2020-01-28:
There is a chance to have rescan as bfamchon said?

PFeci, 2020-05-09:
The official nmcli documentation says (https://developer.gnome.org/NetworkManager/stable/nmcli.html): 
_By default, nmcli ensures that the access point list is no older than 30 seconds and triggers a network scan if necessary. The --rescan can be used to either force or disable the scan regardless of how fresh the access point list is._