friedrith/node-wifi #147: wifi.connect error is undefined when password is incorrect.

kaleem68, 2021-01-26:
## Expected Behavior

wifi.connect should always return a proper error message when the ssid is correct & password is incorrect but it returns the 'undefined'

## Current Behavior
sometimes wifi.connect returns error: Error: Command failed: netsh wlan add profile filename="nodeWifiConnect.xml" when the ssid is correct & password is incorrect, but most of the time error is undefined when ssid is correct and password is incorrect.

## Your Environment
- OS: Windows, Mac

sundong212, 2021-02-23:
Same problem here. The connection would fail at first but without return error messages, and the connection will fallback to the original WiFi since it has the correct profile stored. (Mac)

quanc, 2021-04-06:
Same problem here