friedrith/node-wifi #171: [Bug]: Returning [] empty array on macOS Monterey

juanmartin, 2021-09-23:
### Expected behavior

When scanning networks either by node API using `wifi.scan` or with the CLI using `npx node-wifi-cli --scan`
Should return list of scanned networks.

### Current behavior

When scanning networks either by node API using `wifi.scan` or with the CLI using `npx node-wifi-cli --scan`, it returns `[]` an empty array. No errors, just an empty list.

### Which are the affected features

- [X] node API
- [X] CLI

### Which is your operating system?

MacOS

### Environment

- Macbook Pro 16" Intel Big Sur: working
- Macbook Pro 16" Intel Monterey: empty list
- iMac Intel Monterey: empty list

### Version of node-wifi

2.0.15

### Steps to Reproduce

Scan networks by either CLI or node-wifi api and log the result.

### Solutions

I'm wondering if there are any new permissions to manage the wifi hardware on Monterey.
There were no dialogs shown whatsoever when trying to use the CLI or the node app we are developing that relies on this lib.

Also tried using `debug: true` as a parameter for the `wifi.init` but couldn't see any additional info when logging the resulting networks.

friedrith, 2021-09-25:
The debug `true` is not usable yet.  

Can you try to use command `/System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport --scan`?

juanmartin, 2021-09-27:
Hi! thanks for the follow up.
The command runs fine and shows available networks.

friedrith, 2021-09-27:
I guess the output format is slightly different between Big Sur and Monterey. Can you share an output of the command `node scripts/scan.js`? I suggest you to change few letters in ssid and mac addresses for privacy.

Thank you for your help. Monterey is significant up coming target for the project `node-wifi`.

juanmartin, 2021-09-27:
Sure!

```
node scripts/scan.js
$ /System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport --scan
```
```
                            SSID BSSID             RSSI CHANNEL HT CC SECURITY (auth/unicast/group)
                    XXXX-AMPLIFI                   -81  36      Y  -- WPA2(PSK/AES/AES) 
                    XXXX-AMPLIFI                   -65  36      Y  -- WPA2(PSK/AES/AES) 
                     XXXX-2.4GHz                   -27  11      Y  -- WPA(PSK/TKIP,AES/TKIP) WPA2(PSK/TKIP,AES/TKIP) 
                       XXXX-5GHz                   -33  149     Y  -- WPA(PSK/TKIP,AES/TKIP) WPA2(PSK/TKIP,AES/TKIP)
```

juanmartin, 2021-09-28:
totally empty

friedrith, 2021-09-28:
Ok perfect. I look into your issue.

friedrith, 2021-09-29:
So using your log file I can reproduce the problem in the parser. I see 2 options:
- option 1: I just fix the parser to accept empty bssid
- option 2: I try to find the good paramters to add the bssid in airport command
I would prefer option 2 to keep returning the maximum of information.

Can you tell me if bssid is included in one of the following commands?

```
/System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport scan

/System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -s -x
```

juanmartin, 2021-09-29:
I think you'll be able to do option 2 with the second command, it returns in XML/plist format including bssid.

`airport scan` doesn't return bssid, it's equivalent to `--scan`
[airport-scan.log](https://github.com/friedrith/node-wifi/files/7252666/airport-scan.log)

`airport -s -x` gives xml/plist format with everything
[airport-sx.log](https://github.com/friedrith/node-wifi/files/7252698/airport-sx.log)

Hope it helps. Thank you!

friedrith, 2021-09-29:
Yes it helps a lot. Besides using xml is safer than parsing the command output. It was in the roadmap but I guess we just have to reprioritize this feature. I will work on it.

friedrith, 2021-10-19:
Oups. I started but it is not finished yet. I might be able to push you a version for wednesday evening PT.

friedrith, 2021-10-21:
I am struggling to include refactorizations I was working on for few months. Is it possible to try this alpha: https://www.npmjs.com/package/node-wifi/v/2.1.0-alpha.1 on Big Sur. 

It doesn't include the fix for Monterey but includes refactorizations I might include in the fix for Monterey.

closed 2022-01-07 by juanmartin