friedrith/node-wifi #173: [Bug]: Linux Disconnect Not Working

SuspensionPoint, 2021-11-22:
### Expected behavior

The ReadMe specifies that you must set the interface to null at least.

It then shows that once you set the wifi iface to null, you can then call `disconnect()` method.

On Linux, I'm setting the iface to null, then calling disconnect on the wifi object. 

I expect that the network will be disconnected.

### Current behavior

Instead, whenever I'm calling `disconnect()` on the wifi object I get the following error:

`Error: No interface specified.`

Does this not work if an interface was set to null? I cannot find any mention of this in the readme or issues.

Thanks

### Which are the affected features

- [X] node API
- [ ] CLI

### Which is your operating system?

Linux

### Environment

Ubuntu 18.04.5 LTS

### Version of node-wifi

2.0.14

### Steps to Reproduce

Follow the read me example, then try calling the disconnect method.

### Solutions

Perhaps I'm missing a step, but I'm a little lost here.

friedrith, 2021-11-29:
Sorry for the delay. Can you write the code you are using?

potatolooper, 2022-07-27:
I have the same problem in Ubuntu 20.04.
I try to use a disconnect() and give me the same error
[image]

i dont know if you need more things

SuspensionPoint, 2022-07-27:
Hey there, sorry for the slow response. I did eventually fix this issue. I'm trying to recall what the exact fix was.

I believe I was encountering an issue where when a new network connection was established, for some reason the `iface` for the `this.wifi` instance wasn't being updated to use the new connections `iface`.

Whenever the new connection was made, I manually would call `wifi.init()` and pass in the iface for the new connection.

EllyssonMike, 2022-08-21:
I have the same problem on Debian 10.