friedrith/node-wifi #174: [Feat]: More information about `.init()`

juanmartin, 2022-01-07:
### Is your feature request related to a problem? Please describe.

I need to manipulate WiFi connections and send HTTP requests to devices in the network. On Windows, when having multiple interfaces (eg. Ethernet and WiFi through a USB dongle), these requests fail. As soon as I disable one of the interfaces, they work.
I've tested this with Postman to discard a problem in my implementation, confirming that having only one network enabled fixes the issue and the request goes through.

### Describe the solution you'd like

I'd like to know what other options other than `null` we have in the `iface` option when doing `.init()`.

AFAIK, all we know is this:
```
// Initialize wifi module
// Absolutely necessary even to set interface to null
wifi.init({
  iface: null // network interface, choose a random wifi interface if set to null
});
```
It does the job, but maybe for some edge case, or to have more control, it'd be nice to have at least a list of options depending on OS.

### Describe alternatives you've considered

I've tried putting things like `wlan` or `wlan0` for macOS, and `Wi-Fi` for Windows. Only the Windows one worked but haven't noticed any changes to when it was `null`. 

### Teachability, Documentation, Adoption, Migration Strategy

Nothing really to migrate here, just adding documentation on the options we have for this parameter.
What happens when having a VPN setup for example, which adds another interface? Does this affect `node-wifi`'s choosing of `iface`? 
Putting `null` and hoping that a random interface is the correct one seems a bit odd to me.

If I could please be enlightened on the matter I'd really appreciate it!!

Cheers and happy new year.

friedrith, 2022-01-18:
Hi, happy new year. You are right I will try to improve the documentation. The problem about init is coming from different OS features. Some OS have very simple interface names (like Linux). macOS have simple ones too but a lot and doesn't always take the arguments into account.

juanmartin, 2022-01-20:
Thanks
I ended up using the native OS commands to `grep` the correct interface and then some regex to get the name and force those instead of using `null` as I was having some inconsistencies between different macs.
For mac: `networksetup -listnetworkserviceorder | grep "Wi-Fi"`
For Windows (powershell): `Get-NetAdapter | where name -eq "Wi-Fi" | ft -hidetableheaders`