friedrith/node-wifi #180: [Bug]: Get empty BSSID string in macOS Monterey (> 12.1)

pf895929, 2022-03-28:
### Is your feature request related to a problem? Please describe.

I have an issue when trying to get the BSSID from getCurrentConnection. As a new Apple security update since macOS Monterey 12.1, we are not able to get the BSSID from /System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport without sudo.

### Describe the solution you'd like

Add a flag when running getCurrentConnection to use under sudo. If the app has right privileges, shall return the BSSID; otherwise will thrown an error that needs upper privileges while using it with the sudo flag.

### Describe alternatives you've considered

There is no other way to get the BSSID without sudo.

wobsoriano, 2022-04-03:
Mac and BSSID will not return without sudo privileges now.

keypuncherlabs, 2022-10-25:
Any updates on this?  Seems like a very useful feature to have this implemented or fixed.