friedrith/node-wifi #184: [Bug]: Wrong data returns for getCurrentConnections on windows 11

mussacharles60, 2022-11-11:
### Expected behavior

I have tested on windows 11 to get the current wifi connections but wrong data returns probably indexing the rows from exec results

OS Name:                   Microsoft Windows 11 Pro
OS Version:                10.0.22000 N/A Build 22000

I tried to check directly via command line 

```ssh
C:\Users\Mussa Charles>Netsh WLAN show interfaces

There is 1 interface on the system:

    Name                   : WiFi
    Description            : Broadcom BCM43142 802.11 bgn Wi-Fi M.2 Adapter
    GUID                   : e6f4bd5d-2fd3-452e-afdf-568c8a71b0ce
    Physical address       : 44:1c:a8:9b:94:69
    Interface type         : Primary
    State                  : connected
    SSID                   : Imperial Innovation Super
    BSSID                  : 60:a4:b7:96:5e:52
    Network type           : Infrastructure
    Radio type             : 802.11n
    Authentication         : WPA2-Personal
    Cipher                 : CCMP
    Connection mode        : Auto Connect
    Band                   : 2.4 GHz
    Channel                : 7
    Receive rate (Mbps)    : 72
    Transmit rate (Mbps)   : 72
    Signal                 : 86%
    Profile                : Imperial Innovation Super

    Hosted network status  : Not available
```

### Current behavior

With `getCurrentConnections ` method this is what I get

```js
const wifi = require('node-wifi');
wifi.init({ iface: null });
wifi.getCurrentConnections((error, currentConnections) => {
// ...
});

// results
[
  {
    iface: 'WiFi',
    ssid: 'connected',
    bssid: 'Imperial Innovation Super',
    mac: 'Imperial Innovation Super',
    mode: '60:a4:b7:96:5e:52',
    channel: NaN,
    frequency: NaN,
    signal_level: -64,
    quality: 72,
    security: '802.11n',
    security_flags: 'WPA2-Personal'
  }
]
```
As you can see from `ssid` to  `mode` there are wrong data compared to the directly get from command line interface.

### Which are the affected features

- [X] node API
- [X] CLI

### Which is your operating system?

Windows

### Environment

electron

### Version of node-wifi

2.0.16

### Steps to Reproduce

Install  node-wifi v2.0.16

`npm i node-wifi@2.0.16`

Import node-wifi to project

```js
const wifi = require('node-wifi');
wifi.init({ iface: null });

// get current connections
wifi.getCurrentConnections((error, currentConnections) => {
// ...
});
```

### Solutions

It should be fixed by indexing the correct data from exec results, like at the expected behavior

JordanPFVR, 2023-06-13:
Still appears to be an issue.

eltoroit, 2023-09-01:
The problem is that the code is assuming the fields are always shown in the same order. [This function](https://github.com/friedrith/node-wifi/blob/1a4364e67c9789555a00656d35287f0cebb3add8/src/windows-current-connections.js#L5) would need to be rewritten in a more generic form.

eltoroit, 2023-09-02:
I have proposed a change to the code that fixes this issue.