friedrith/node-wifi #189: I can't get wifi list when I use wifi.scan().

kamelorac, 2023-05-23:
### Expected behavior

If I use wifi.scan(), I will get wifi list.
networks = [
            {
              ssid: '...',
              bssid: '...',
              mac: '...', // equals to bssid (for retrocompatibility)
              channel: <number>,
              frequency: <number>, // in MHz
              signal_level: <number>, // in dB
              quality: <number>, // same as signal level but in %
              security: 'WPA WPA2' // format depending on locale for open networks in Windows
              security_flags: '...' // encryption protocols (format currently depending of the OS)
              mode: '...' // network mode like Infra (format currently depending of the OS)
            },
            ...
        ];

### Current behavior

If I use wifi.scan(), I get empty wifi list.
networks = [];

### Which are the affected features

- [ ] node API
- [ ] CLI

### Which is your operating system?

Linux

### Environment

Ubuntu 22.0.4

### Version of node-wifi

2.0.16

### Steps to Reproduce

I am working on electron project.
...
const WiFiControl = require('node-wifi');

WiFiControl.init({
  iface: null,
});
const Allnetworks: any[] = [];

WiFiControl.scan((error, networks) => {
  if (error) {
    console.log('*************');
    console.log(error);
  } else {
    console.log('8888888888888888');
    Allnetworks = networks;
  }
});
.....
console.log(Allnetworks); // empty []

### Solutions

How to solve it?

AM-51, 2024-08-05:
Same problem on macbook pro m1, sonoma 14.5

vuvivian, 2024-10-28:
Do you have a solution?

55Cancri, 2024-12-09:
this package uses macos airport command. If you run through terminal on later versions of macos, it won't work because its been deprecated: https://www.intuitibits.com/2024/03/14/goodbye-airport/