friedrith/node-wifi #195: Unable to detect SSID after update to MacOS Sonoma 14.4 (23E214)

akanksha-atrey, 2024-03-29:
### Expected behavior

The package was working as expected. 

### Current behavior

After updating to MacOS Sonoma 14.4, the package is not detecting any networks.

### Which are the affected features

- [X] node API
- [ ] CLI

### Which is your operating system?

MacOS

### Environment

MacOS Sonoma 14.4 (23E214)

### Version of node-wifi

2.0.16

### Steps to Reproduce

```
try {
    await wifi.init({ iface: null });
    const networks = await wifi.scan();

    console.log(`Found the following SSIDs: ${networks}`)

    return networks;
} catch (error) {
    console.error('Error scanning networks:', error);
}
```

### Solutions

Is this a bug or a local issue?

iogi, 2024-04-18:
The same problem occurred.
This is caused by the airport command used by node-wifi being deprecated in MacOS Sonoma.
https://github.com/friedrith/node-wifi/blob/1a4364e67c9789555a00656d35287f0cebb3add8/src/macOS/scan/command.js#L2

akanksha-atrey, 2024-04-18:
Yes, I realized that. I figured out a hack using pyobjc modules for CoreWLAN/CoreLocation. Not a pretty way of doing it, but does the job for the time being.

shaguarger, 2024-04-25:
I did some research at lunch and found a hint here:
https://github.com/friedrith/node-wifi/issues/195#issuecomment-2063874988

Then I found this:
https://stackoverflow.com/questions/78153143/assistance-in-extracting-fields-and-formatting-using-pyobjc-and-corewlan

Maybe there is a way to use this here?

aljaramillo, 2024-06-11:
I just wanted to chime in and mention that I am also experiencing this issue on my end. After updating to MacOS Sonoma 14.4 (23E214), my system is unable to detect the SSID. Any insights or workarounds would be greatly appreciated.

deanylev, 2024-10-14:
Obviously trickier but I managed to work around this by writing a native module that calls this method https://developer.apple.com/documentation/corewlan/cwinterface/scanfornetworks(withssid:)?language=objc

It's blocking so be sure to wrap it in a promise/callback, I used a `Napi::Promise`

shaguarger, 2024-11-07:
FYI: There is a way with system profiler. We did a PR in another repo.

https://github.com/sebhildebrandt/systeminformation/pull/937